ljalves/linux_media #151: TBS5281 card DVB-C network: invalid checksum problem

schneci, 2017-01-24:
I'm using a TBS5281 card with Tvheadend 4.2 for receiving DVB-C channels and the Silicon Labs Si2168 opensource TBS driver at most of the channels generate the following error messages in log file of Tvheadend:

> TS: Silicon Labs Si2168 : DVB-C #0/DVB-C network/250MHz/TV2: MPEG2VIDEO @ #201 Continuity counter error (total 1557)
TS: Silicon Labs Si2168 : DVB-C #0/DVB-C network/250MHz/TV2: MPEG2AUDIO @ #210 Continuity counter error (total 252)
TS: Silicon Labs Si2168 : DVB-C #0/DVB-C network/250MHz/TV2: TELETEXT @ #220 Continuity counter error (total 191)
TS: Silicon Labs Si2168 : DVB-C #0/DVB-C network/250MHz/TV2: MPEG2AUDIO @ #211 Continuity counter error (total 236)
TS: Silicon Labs Si2168 : DVB-C #0/DVB-C network/250MHz/TV2: MPEG2VIDEO @ #201 Corrupted PES header (errors 1)
pmt: 250MHz in DVB-C network: invalid checksum (len 55, errors 1)
eit: 250MHz in DVB-C network: invalid checksum (len 1718, errors 40)

If I use the proprietary TBS driver everything works correctly. Could somebody help to me.

LubosD, 2017-01-30:
I am seeing a similar problem on a TBS6985 (quad DVB-S) card. I use VDR and after the card has been in use for a long time (days?), the picture starts breaking up. Restarting VDR always fixes the problem. I've never seen this with official drivers.

schneci, 2017-01-30:
My problem has solved by using the following options in Libreelelc 7.95.1 version which uses 4.9x kernel version:

`options cx231xx transfer_mode=0`

In Libreelec or Openelec environment you can set this option using following command:

`echo "options cx231xx transfer_mode=0" > /storage/.config/modprobe.d/cx231xx.conf`

Please see the following link:
[https://github.com/crazycat69/linux_media/issues/17#event-936963251](url)

closed 2017-03-26 by schneci