ljalves/linux_media #152: V4L drivers

mathojojo, 2017-02-26:
Hello,

Why are those TBS drivers not included in the V4L drivers ? Wouldn't it be easier to install ?

each time my kernel is updated, I need to re-build/make/make install the proprietary driver. I thought that it would be easier with the opensource drivers.
But it looks like we will have the same problem, if they are not available in V4L, or at least, in a ppa repository.

Thank you :)

lukastribus, 2017-04-15:
I don't disagree, and upstreaming those drivers would eliminate all those build issues and simplify things for a lot of people, but I guess its a non-trivial task to upstream all those drivers.

dropz-one, 2017-10-03:
Hello how could we help to add drivers to mainstream ?