ljalves/linux_media #153: TBS 6909 Kernel 4.4.0-71 - TBSECP3 frontend 0 attach failed.

nightshadow36, 2017-04-03:
Help please.
Just updated my kernel and updated my card drivers now i get TBSECP3 frontend attach failed.
I dont know whats going on it says registering and failed on the same numbers but not showing 32 tuners?
What to do thank you.

[ 36.502707] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 36.579398] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 36.700700] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 2 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 36.796576] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 3 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 36.916593] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 4 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 37.028659] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 5 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 37.109126] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 6 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 37.230788] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 7 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 62.453879] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 8 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 62.552673] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 9 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 62.625264] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 10 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 62.695869] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 11 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 62.799870] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 12 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 62.919852] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 13 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 62.988065] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 14 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 63.084404] TBSECP3 driver 0000:07:00.0: DVB: registering adapter 15 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
[ 63.209636] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 0 attach failed
[ 63.209637] TBSECP3 driver 0000:09:00.0: frontend attach failed
[ 63.365895] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 1 attach failed
[ 63.365896] TBSECP3 driver 0000:09:00.0: frontend attach failed
[ 63.481883] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 2 attach failed
[ 63.481884] TBSECP3 driver 0000:09:00.0: frontend attach failed
[ 63.593715] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 3 attach failed
[ 63.593716] TBSECP3 driver 0000:09:00.0: frontend attach failed
[ 63.753893] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 4 attach failed
[ 63.753894] TBSECP3 driver 0000:09:00.0: frontend attach failed
[ 63.905860] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 5 attach failed
[ 63.905861] TBSECP3 driver 0000:09:00.0: frontend attach failed
[ 64.025849] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 6 attach failed
[ 64.025850] TBSECP3 driver 0000:09:00.0: frontend attach failed
[ 64.185843] TBSECP3 driver 0000:09:00.0: TBSECP3 frontend 7 attach failed
[ 64.185844] TBSECP3 driver 0000:09:00.0: frontend attach failed
[ 64.297841] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 0 attach failed
[ 64.297842] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[ 64.453818] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 1 attach failed
[ 64.453819] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[ 64.561545] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 2 attach failed
[ 64.561546] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[ 64.685385] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 3 attach failed
[ 64.685387] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[ 64.841859] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 4 attach failed
[ 64.841860] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[ 64.961833] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 5 attach failed
[ 64.961834] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[ 65.073801] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 6 attach failed
[ 65.073802] TBSECP3 driver 0000:0a:00.0: frontend attach failed
[ 65.185676] TBSECP3 driver 0000:0a:00.0: TBSECP3 frontend 7 attach failed
[ 65.185677] TBSECP3 driver 0000:0a:00.0: frontend attach failed

welshdemon, 2017-04-27:
I suggest install older kernel for now if you need the card working quickly. Then maybe reinstall drivers

twentythousandphantoms, 2017-07-08:
My 6909 cards registered at the same seconds:
1st card - 36th second,
2nd card - 61th second.

And others people 6909-cards (on few different forums) shows the same results
Its too late. 

For exapmle, 6985 registered at 11th second. 

How to make the 6909's start faster?

```
# uname -r
3.19.0-80-generic
```

crazycat69, 2017-07-08:
Mxl584 tuner/demod require some big firmware upload on start, so need some time :-)

momomo, 2017-07-25:
@crazycat69  I am having similar issues I believe: 

https://tvheadend.org/boards/5/topics/27406

 I am not so well oriented in this field. Care to explain what is the best fix for this?