ljalves/linux_media #154: Compile error after changing 'make menuconfig' options

IanPHeitmans, 2017-04-10:
When I follow the default steps to compile the ljalves drivers, my TBS6982 card is not recognized.

uname -a :
`Linux voyage 4.1.6-voyage #1 SMP Fri Jun 3 10:04:48 HKT 2016 i686 GNU/Linux`

lspci | grep -i saa :
`13:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)`

sudo modprobe saa716x_budget int_type=1 :
`modprobe: FATAL: Module saa716x_budget not found.`

So I activated

> Support for SAA716x family from NXP/Philips
> SAA7160/1/2 based Budget PCIe cards (DVB only)

with make menuconfig

But when I 'make' after that change, I get the following error:

```
  CC [M]  /usr/src/media_build/v4l/cxd2820r_core.o
/usr/src/media_build/v4l/cxd2820r_core.c: In function 'cxd2820r_wr_reg_val_mask_tab':
/usr/src/media_build/v4l/cxd2820r_core.c:48:4: error: implicit declaration of function 'regmap_write_bits' [-Werror=implicit-function-declaration]
    ret = regmap_write_bits(regmap, reg, mask, val);
    ^
/usr/src/media_build/v4l/cxd2820r_core.c: In function 'cxd2820r_i2c_gate_ctrl':
/usr/src/media_build/v4l/cxd2820r_core.c:427:21: warning: unused variable 'client' [-Wunused-variable]
  struct i2c_client *client = priv->client[0];
                     ^
cc1: some warnings being treated as errors
scripts/Makefile.build:258: recipe for target '/usr/src/media_build/v4l/cxd2820r_core.o' failed
make[3]: *** [/usr/src/media_build/v4l/cxd2820r_core.o] Error 1
Makefile:1401: recipe for target '_module_/usr/src/media_build/v4l' failed
make[2]: *** [_module_/usr/src/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.1.6-voyage'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/usr/src/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2

```

The error is gone and compilation is succesfull when I disable the option :
`SAA7160/1/2 based Budget PCIe cards (DVB only)`
but then my TBS6982 isn't working.

Is this a bug or am I doing something wrong?

lukastribus, 2017-04-15:
Linux kernel 4.6 is needed for the saa716x_budget module to be build.

IanPHeitmans, 2017-04-16:
Unfortunately I get a completely different error with kernel 4.8.17:

uname -a :
`Linux voyage 4.8.17-voyage #1 SMP Sun Apr 16 00:23:27 GMT 2017 x86_64 GNU/Linux`

```
  LD [M]  /usr/src/media_build/v4l/snd-bt87x.o
make[3]: *** No rule to make target '/usr/src/media_build/v4l/frame_vector.c', needed by '/usr/src/media_build/v4l/frame_vector.o'.  Stop.
Makefile:1473: recipe for target '_module_/usr/src/media_build/v4l' failed
make[2]: *** [_module_/usr/src/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.8.17-voyage'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/usr/src/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2
root@voyage:/usr/src/media_build#

```

jfowkes, 2017-05-13:
I also get exactly the same error on 4.9.16

mfyorp, 2017-06-16:
I have the same error.
I manager to skip the error by copying frame_vector.c
[cp ../media/mm/frame_vector.c v4l/](url)
Warning: I have actually no idea what I'm doing. The above might be a very bad idea and break other things.

jfowkes, 2017-06-18:
Thanks @mfyorp `cp ../media/mm/frame_vector.c v4l/` fixed it for me!