ljalves/linux_media #155: saa716x_budget.ko not created with TBS 6285 (SAA7160 rev 02)

Chez2, 2017-04-12:
This is my first attempt at using the open source driver.

The system is a fresh install of Ubuntu 16.04.2 LTS.

uname -a

	4.4.0-72-generic

sudo lspci -vvvnn | grep --after-context=28 7160

	0d:00.0 Multimedia controller [0480]: Philips Semiconductors SAA7160 [1131:7160] (rev 02)
		Subsystem: Device [6285:0001]
		Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
		Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
		Latency: 0, Cache Line Size: 32 bytes
		Interrupt: pin A routed to IRQ 16
		Region 0: Memory at f7700000 (64-bit, non-prefetchable) [size=1M]
		Capabilities: [40] MSI: Enable- Count=1/32 Maskable- 64bit+
			Address: 0000000000000000  Data: 0000
		Capabilities: [50] Express (v1) Endpoint, MSI 00
			DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s <256ns, L1 <1us
				ExtTag- AttnBtn- AttnInd- PwrInd- RBE- FLReset-
			DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
				RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop+
				MaxPayload 128 bytes, MaxReadReq 128 bytes
			DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
			LnkCap:	Port #1, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
				ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
			LnkCtl:	ASPM Disabled; RCB 128 bytes Disabled- CommClk-
				ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
			LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
		Capabilities: [74] Power Management version 2
			Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot-,D3cold-)
			Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
		Capabilities: [80] Vendor Specific Information: Len=50 <?>
		Capabilities: [100 v1] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>

I've followed the instructions at https://github.com/ljalves/linux_media/wiki/Installating.  The build completes successfully. However I receive the following error when I attempt to load the module:

	modprobe: FATAL: Module saa716x_budget not found in directory /lib/modules/4.4.0-72-generic
		
Is anybody able to help please ?

lukastribus, 2017-04-15:
Linux kernel 4.6 is needed for this driver.

Upgrade to the HWE kernel (currently 4.8) in ubuntu and rebuild:
```
sudo apt-get update
sudo apt-get install --install-recommends linux-generic-hwe-16.04
```

Chez2, 2017-04-16:
I followed the solution which worked for me. Many thanks.

closed 2017-04-16 by Chez2