ljalves/linux_media #156: saa716xx_budget not created

Krobar, 2017-05-08:
Using Ubuntu 15.10 on the the newest 4.2 Kernel I don't seem to get the driver I need. Make output looks quite clean (Below is second run):

rob@TVHeadend:~/media_build/v4l$ ls /lib/modules/4.2.0-42-generic//kernel/drivers/media/pci/saa716x/
saa716x_core.ko  saa716x_ff.ko  saa716x_hybrid.ko
rob@TVHeadend:~/media_build/v4l$ cd ..

rob@TVHeadend:~/media_build$ make
make -C /home/rob/media_build/v4l
make[1]: Entering directory '/home/rob/media_build/v4l'
creating symbolic links...
make -C firmware prep
make[2]: Entering directory '/home/rob/media_build/v4l/firmware'
make[2]: Leaving directory '/home/rob/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/home/rob/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/rob/media_build/v4l/firmware'
Kernel build directory is /lib/modules/4.2.0-42-generic/build
make -C ../linux apply_patches
make[2]: Entering directory '/home/rob/media_build/linux'
Syncing with dir ../../media
Patches for 4.2.0-42-generic already applied.
make[2]: Leaving directory '/home/rob/media_build/linux'
make -C /lib/modules/4.2.0-42-generic/build SUBDIRS=/home/rob/media_build/v4l  modules
make[2]: Entering directory '/usr/src/linux-headers-4.2.0-42-generic'
  Building modules, stage 2.
  MODPOST 656 modules
WARNING: "frame_vector_to_pages" [/home/rob/media_build/v4l/videobuf2-vmalloc.ko] undefined!
WARNING: "frame_vector_to_pfns" [/home/rob/media_build/v4l/videobuf2-vmalloc.ko] undefined!
WARNING: "frame_vector_create" [/home/rob/media_build/v4l/videobuf2-memops.ko] undefined!
WARNING: "frame_vector_destroy" [/home/rob/media_build/v4l/videobuf2-memops.ko] undefined!
WARNING: "get_vaddr_frames" [/home/rob/media_build/v4l/videobuf2-memops.ko] undefined!
WARNING: "put_vaddr_frames" [/home/rob/media_build/v4l/videobuf2-memops.ko] undefined!
WARNING: "frame_vector_to_pages" [/home/rob/media_build/v4l/videobuf2-dma-sg.ko] undefined!
WARNING: "frame_vector_to_pages" [/home/rob/media_build/v4l/videobuf2-dma-contig.ko] undefined!
WARNING: "frame_vector_to_pfns" [/home/rob/media_build/v4l/videobuf2-dma-contig.ko] undefined!
make[2]: Leaving directory '/usr/src/linux-headers-4.2.0-42-generic'
./scripts/rmmod.pl check
found 656 modules
make[1]: Leaving directory '/home/rob/media_build/v4l'

Try to use this with a TBS6985, it compiled on the same system with code from May2015 a few years ago. Since upgraded the kernel (With headers) and now cant seem to get it to detect the card. Any help appreciated.

Krobar, 2017-05-09:
Ignore the above, it needs a newer kernel. 4.2 fails but 4.8 is ok.

mochour, 2017-05-19:
Hi,

I have the same problem but on Debian Jessie kernel 3.16.43-2. What is the resolution, when I don't want to upgrade kernel?

I'v followed the install instructions, but the result is always same the driver is not build. 
My card is TBS6280 DVB-T2.

Thanks
M

crazycat69, 2017-05-19:
6280 require cxd2820r driver, so need at least 4.4 kernel.
install debian 9 with 4.9 kernel

mochour, 2017-05-21:
Ok, I've upgraded the kernel to 4.9 and the build was successful. 

I can see a lot of continuity errors for one multiplex (others are ok).  When I tune the multiplex directly on TV, there are no artifacts or visible problems. 

Do you have any hint what to check? Can it be problem of DVB API?

Thanks

crazycat69, 2017-05-21:
What muxes you lock ? DVB-T, T2 or C ?

mochour, 2017-05-21:
DVB-T 

Current parameters:
    Frequency:  778000.000 kHz
    Inversion:  OFF
    Bandwidth:  8 MHz
    Stream code rate (hi prio):  FEC 3/4
    Stream code rate (lo prio):  FEC 1/2
    Modulation:  QAM 64
    Transmission mode:  8k mode
    Guard interval:  1/8
    Hierarchy:  none

crazycat69, 2017-05-21:
Try drivers from my media_build
https://bitbucket.org/CrazyCat/media_build

This forked from this repo
https://github.com/crazycat69/linux_media

mochour, 2017-05-21:
I think, I used it ... according to install guide:
`git clone https://github.com/ljalves/media_build.git
git clone --depth=1 https://github.com/ljalves/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make distclean
make`

mochour, 2017-05-21:
I am getting much better results when using your driver than proprietary TBS drivers, but this is only problem that bothers me :-(.

Krobar, 2017-05-22:
I get this continuity errors at times on 2/4 tuners on a 6985, slightly crappy hardware I think. TBS claim it is a motherboard compatibility, have tried 2 completely different motherboards and still get the problem, if it is a compatibility issues it seems common enough that other cards don't suffer the problem.

mochour, 2017-05-22:
I think, you are right. Before reinstall, I had similar problem as now, sometimes could not tune two channels from one multiplex. The problem is gone, but continuity error appeared. 

Do you know how to use verbose parameter for the driver?

eldonT, 2017-08-22:
is it the same problem (saa716x_budget not built) for TBS 6922 on a 4.4.0 kernel (ubuntu trusty) ?

i did see the cxd2820r 4.6 kernel limitation during make but the 6922 does not use that one so i was hoping it would work.. 

i'm trying to build again from scratch, as i had a perl error when executing : 
make dir DIR=../media
saying "device out of space", while in fact my /usr/src is simply mounted on a different filesystem.

the second build also "skipped" saa716x_budget, let me know if there's a way to use it on a 4.4 kernel
thx