ljalves/linux_media #157: TBS 6984 install - showing errors

speedyrazor, 2017-05-13:
Hi, I am installing a TBS 6984 in Ubuntu 17.04 server, i am using the linux firmware tbs-linux-drivers_v170330.zip, copied to /lib/firmware/, done the install as per instructions, but when running dmesg | grep frontend, I am seeing the below, is this correct, or is something going wrong?

$ dmesg | grep frontend
[   11.787058] i2c i2c-1: cx24117: Attaching frontend 0
[   11.805156] SAA716x Budget 0000:0b:00.0: TurboSight TBS 6984 frontend 0 doesn't seem to have a isl6422b on the i2c bus.
[   11.805158] SAA716x Budget 0000:0b:00.0: DVB: registering adapter 0 frontend 0 (Conexant CX24117/CX24132)...
[   11.805559] i2c i2c-1: cx24117: Attaching frontend 1
[   11.810766] SAA716x Budget 0000:0b:00.0: TurboSight TBS 6984 frontend 1 doesn't seem to have a isl6422b on the i2c bus.
[   11.810767] SAA716x Budget 0000:0b:00.0: DVB: registering adapter 1 frontend 0 (Conexant CX24117/CX24132)...
[   11.810994] i2c i2c-0: cx24117: Attaching frontend 0
[   11.816137] SAA716x Budget 0000:0b:00.0: TurboSight TBS 6984 frontend 2 doesn't seem to have a isl6422b on the i2c bus.
[   11.816138] SAA716x Budget 0000:0b:00.0: DVB: registering adapter 2 frontend 0 (Conexant CX24117/CX24132)...
[   11.816353] i2c i2c-0: cx24117: Attaching frontend 1
[   11.821491] SAA716x Budget 0000:0b:00.0: TurboSight TBS 6984 frontend 3 doesn't seem to have a isl6422b on the i2c bus.
[   11.821492] SAA716x Budget 0000:0b:00.0: DVB: registering adapter 3 frontend 0 (Conexant CX24117/CX24132)...
[   13.695035] SAA716x Budget 0000:0b:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (950000..2150000)
[   17.223296] SAA716x Budget 0000:0b:00.0: DVB: adapter 2 frontend 0 frequency 0 out of range (950000..2150000)
[   18.483163] SAA716x Budget 0000:0b:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (950000..2150000)
[   18.711174] SAA716x Budget 0000:0b:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (950000..2150000)

Kind regards.

ljalves, 2017-05-13:
All seems ok.

speedyrazor, 2017-05-14:
OK cool, thanks. Also, in the installation procedure on here, it says to install the firmware file tbs-linux-drivers_v130901.zip, but the latest is tbs-linux-drivers_v170330.zip. Which one should I use, does it matter?

crazycat69, 2017-05-14:
This drivers, not firmware :-) And this tbs closed-source driver, discontinued and not supported now. Install open-source drivers instead.8:53, 14 мая 2017 г., speedyrazor <notifications@github.com>:OK cool, thanks. Also, in the installation procedure on here, it says to install the firmware file tbs-linux-drivers_v130901.zip, but the latest is tbs-linux-drivers_v170330.zip. Which one should I use, does it matter?

speedyrazor, 2017-05-14:
Hmm, now I am getting this error:

[23188.174511] i2c i2c-0: cx24117_load_firmware() Error running FW.
[23188.174527] i2c i2c-0: cx24117_firmware_ondemand: Writing firmware failed
[23188.174573] i2c i2c-0: cx24117_load_firmware() Error running FW.
[23188.174588] i2c i2c-0: cx24117_firmware_ondemand: Writing firmware failed

A reboot fixes it, but waiting a while results in this error message again. Any idea whats going on?

crazycat69, 2017-05-14:
You need cx24117 firmware **dvb-fe-cx24117.fw** .  TBS provide it in archive http://www.tbsdtv.com/download/document/linux/tbs-tuner-firmwares_v1.0.tar.bz2

P.S. TBS DTV open-source drivers, fork from this repo - https://github.com/tbsdtv/linux_media

speedyrazor, 2017-05-14:
I have installed the firmware file, /lib/firmware/dvb-fe-cx24117.fw, it runs fine for a while then give me the above errors.
Would you recommend this fork from this repo - https://github.com/tbsdtv/linux_media ?

crazycat69, 2017-05-14:
For TBS6984 is same :)