ljalves/linux_media #158: Help needed for noob installing Avermedia TV (phillips SAA7160) card on Ubuntu

robert9422, 2017-05-27:
Hi,
Thanks for all the work so far, would like to learn more about Ubuntu.

I can't seem to get my Computer to use my TV card.

A little about my system,
-Computer-Processor: 4x Intel(R) Core(TM) i5-2400 CPU @ 3.10GHz
Memory: 4025MB (1457MB used)
Operating System: Ubuntu 17.04
User Name: rob (rob)
Date/Time: Sat 27 May 2017 11:04:46 AEST
-Display-Resolution	: 1280x720 pixels
OpenGL Renderer	: GeForce GT 610/PCIe/SSE2
X11 Vendor: The X.Org Foundation
-Multimedia-Audio Adapter: HDA-Intel - HDA Intel PCH
Audio Adapter	: HDA-Intel - HDA NVidia

-PCI Devices-

Multimedia controller: Philips Semiconductors SAA7160 (rev 01) *****THIS IS THE PROBLEM*******
  

I followed the instructions on the previous page and did not get any error messages, but when I start my tvheadend backend there are is no DVB input.

Any ideas?

Thanks in Advance,
Regards,
Rob

crazycat69, 2017-05-27:
Your DVB card not supported in linux V4L DVB drivers.

robert9422, 2017-05-27:
Bugger.

closed 2019-03-04 by robert9422