ljalves/linux_media #159: TBS 6281 SE Problem

Simiil, 2017-05-28:
i compiled and installed the driver and modules using this guide: https://www.linuxtv.org/wiki/index.php/TBS_driver_installation

strangely enough, lspci shows the card as 

    01:00.0 Multimedia controller: TBS Technologies DVB-S2 4 Tuner PCIe Card
        Subsystem: Device 6281:0002
        Flags: fast devsel, IRQ 11
        Memory at f7d00000 (32-bit, non-prefetchable) [disabled] [size=256K]
	Capabilities: [50] Power Management version 3
	Capabilities: [70] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [90] Express Endpoint, MSI 00
	Capabilities: [100] Device Serial Number 00-00-00-00-00-00-00-00

Even though it is not a DVB-S2, 4Tuner card. 

even after modprobing, lspci does not indicate that the card is used by the module. `/sys/class/dvb/` exists, but contains nothing and `/dev/dvb` does not exist at all.

dmesg only says (after modprobing)

    [  168.681186] media: Linux media interface: v0.10
    [  168.685066] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	989e2fe3c894330ede73c44f98e431c68a0f2824 tbsecp3: Added TBS6281SE

and lsmod:

    # lsmod | grep -i saa
    saa716x_budget         36864  0
    tas2101                24576  1 saa716x_budget 
    cx24117                32768  1 saa716x_budget
    saa716x_core           61440  1 saa716x_budget
    dvb_core              110592  2 saa716x_budget,saa716x_core

using the official drivers works.

i am running Archlinux with a 4.10 kernel.

ObviusOwl, 2017-06-17:
I can confirm this issue. I'm using mint 17.3 (ubuntu 14.04 LTS).

hei-pa, 2017-08-29:
I tried the AUR Package from Arch linux and there are the same messages. Any update on this?

vassilit, 2018-03-06:
I've got the same issue, but strangely enough, it used to work with exactly the same drivers on another motherboard.
(I moved a system HDD from one motherboard to another)
Maybe a problem with MSI/IRQ.