ljalves/linux_media #161: Error - "incompatible pointer type" Fedora 26 Kernel 4.11.8

TheYetiWakes, 2017-08-05:
Hi
Receive the following error during make on above system:

`/xxxx/xxxx/media_build/v4l/videobuf-dma-sg.c:456:11: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .fault = videobuf_vm_fault,
           ^~~~~~~~~~~~~~~~~
`

mfyorp, 2017-09-17:
I have the exact same error on Gentoo with kernel 4.12.12

cz172638, 2017-09-19:
maybe check fixes in kernel for f26/f27 as in f27 landed gcc7

mfyorp, 2017-09-20:
The error remains the same, both with gcc 5.4.0 and gcc 7.2.0. Both on kernel 4.12.12.

alanswanson, 2017-09-24:
Development appears to have switched to the semi-official forks (supported by TBS for a few of their own cards after the great work by ljalves and co) at;

https://github.com/tbsdtv/linux_media
https://github.com/tbsdtv/media_build

And this is why you should upstream drivers otherwise you continually need to sync with newer kernel releases. This fork does appear to have a target to upstream into the mainline kernel.