ljalves/linux_media #164: EVOLVEO SIGMA T2

kapitanf, 2017-09-26:
Hello, is there support for EVOLVEO SIGMA T2 USB Stick? VID 0572 PID C68A (aka MYGICA_T230C2). "sudo scan cz-All -o zap" output:

'
ERROR: cannot parse'[CHANNEL]
'
ERROR: cannot parse'	DELIVERY_SYSTEM = DVBT
'
ERROR: cannot parse'	FREQUENCY = 570000000
'
ERROR: cannot parse'	BANDWIDTH_HZ = 8000000
'
ERROR: cannot parse'	CODE_RATE_HP = 2/3
'
ERROR: cannot parse'	CODE_RATE_LP = NONE
'
ERROR: cannot parse'	MODULATION = QAM/64
'
ERROR: cannot parse'	TRANSMISSION_MODE = 8K
'
ERROR: cannot parse'	GUARD_INTERVAL = 1/8
'
ERROR: cannot parse'	HIERARCHY = NONE
'
ERROR: cannot parse'	INVERSION = AUTO
'
ERROR: initial tuning failed
dumping lists (0 services)
Done.

"w_scan -c CZ" output:

554000: (time: 00:20.101) 
562000: (time: 00:22.101) 
570000: (time: 00:24.101)         signal ok:	QAM_AUTO f = 570000 kHz I999B8C999D999T999G999Y999 (0:0:0)
        Info: no data from PAT after 2 seconds
        deleting (QAM_AUTO f = 570000 kHz I999B8C999D999T999G999Y999 (0:0:0))
578000: (time: 00:27.462) 
586000: (time: 00:32.012) 
594000: (time: 00:34.022) 

Any idea to solve this? Thanks

kapitanf, 2017-09-26:
And "scan -c" output shows:

using '/dev/dvb/adapter0/frontend0' and '/dev/dvb/adapter0/demux0'
WARNING: filter timeout pid 0x0011
WARNING: filter timeout pid 0x0000
dumping lists (0 services)
Done.

kapitanf, 2017-10-05:
Here is complete output of "dvbv5-scan /usr/share/dvb/dvb-t/cz-All":
[http://185.33.146.234/codes/dvbv5scan-1.txt](url)

crazycat69, 2017-10-05:
Driver loaded, but signal level pretty low and SNR is bad ( so you not receive any stream for locked muxes). Check your device on Windows. Or Android with PadTV app.

Pbk2, 2017-10-08:
I can confirm the usb stick EVOLVEO SIGMA T2 (0572:c68a) does not work. Dmesg does not show any errors. Tuner and demodulator is properly attached and firmwares downloaded. It seems the tuner is working but no TS data stream is received.

Problem is that this device must have set dd_ts_mode to this configuration (driver si2168.c):

```
/* set ts mode to parallel,clock manual*/
memcpy(cmd.args, "\x14\x00\x01\x10\x26\x00", 6);
cmd.wlen = 6;
cmd.rlen = 4;
ret = si2168_cmd_execute(client, &cmd);
if (ret)
	goto err;

/* set ts freq to 10Mhz*/
memcpy(cmd.args, "\x14\x00\x0d\x10\xe8\x03", 6);
cmd.wlen = 6;
cmd.rlen = 4;
ret = si2168_cmd_execute(client, &cmd);
if (ret)
	goto err;

/* set ts parallel prop*/
memcpy(cmd.args, "\x14\x00\x09\x10\xe3\x08", 6);
cmd.wlen = 6;
cmd.rlen = 4;
ret = si2168_cmd_execute(client, &cmd);
if (ret)
	goto err;
```

The same settings is used on Win7 (according to usb dump). If this configuration is necessary for the device 0572:c68a only, I don't know.

basco, 2018-03-04:
@Pbk2 
in which function inside si2168.c should this be added?

kapitanf, 2018-07-01:
Thanks Pbk2

kapitanf, 2018-07-01:
Thanks Pbk2

raslal, 2019-06-17:
we're almost there [(code)](https://patchwork.kernel.org/patch/10997259/) :-)

tbzatek, 2020-05-25:
(posting on behalf of my colleague Milan Crha):

Tested with kernel 5.6.12 (on Fedora 32). The USB stick is properly recognized, but w_scan doesn't find any channels, neither on DVB-T nor DVB-T2. Only after the firmware is added, as describe here [1], the w_scan starts to work as expected (just add the firmware files, unplug the device and plug it again).

If you happen to have tvheadend not to search for DVB-T2 channels, then maybe you'd need to update the channel table, similarly as suggested here [2]. That depends on the country you use.

[1] https://www.linuxtv.org/wiki/index.php/Geniatech_T230C#Firmware
[2] https://bugzilla.redhat.com/show_bug.cgi?id=1839363

Rohlik, 2020-09-19:
@tbzatek I also have this stick and Debian 10 with 5.5.0-0.bpo.2-amd64 kernel but even after I downloaded the firmware files and stored them in `/lib/firmware` and reboot OS, I was not successful with `w_scan`. No channels found.

**dmesg**:
```
[   58.497205] si2168 20-0064: Silicon Labs Si2168-D60 successfully identified
[   58.497206] si2168 20-0064: firmware version: D 6.0.1
[   58.500551] si2157 21-0060: Silicon Labs Si2141 successfully attached
[   58.500572] usb 2-1: DVB: registering adapter 0 frontend 0 (Silicon Labs Si2168)...
[   58.500576] dvbdev: dvb_create_media_entity: media entity 'Silicon Labs Si2168' registered.
[   58.500805] Registered IR keymap rc-total-media-in-hand-02
[   58.500822] rc rc0: MyGica Mini DVB-T2 USB Stick T230C v2 as /devices/pci0000:00/0000:00:13.2/usb2/2-1/rc/rc0
[   58.500846] rc rc0: lirc_dev: driver dvb_usb_dvbsky registered at minor = 0, scancode receiver, no transmitter
[   58.500878] input: MyGica Mini DVB-T2 USB Stick T230C v2 as /devices/pci0000:00/0000:00:13.2/usb2/2-1/rc/rc0/input29
[   58.500907] usb 2-1: dvb_usb_v2: schedule remote query interval to 300 msecs
[   58.500909] usb 2-1: dvb_usb_v2: 'MyGica Mini DVB-T2 USB Stick T230C v2' successfully initialized and connected
[   71.957723] si2168 20-0064: firmware: direct-loading firmware dvb-demod-si2168-d60-01.fw
[   71.957726] si2168 20-0064: downloading firmware from file 'dvb-demod-si2168-d60-01.fw'
[   71.957728] si2168 20-0064: firmware download failed -22
[   71.974380] si2157 21-0060: found a 'Silicon Labs Si2141-A10'
[   71.974800] si2157 21-0060: firmware: direct-loading firmware dvb-tuner-si2141-a10-01.fw
[   71.974802] si2157 21-0060: firmware file 'dvb-tuner-si2141-a10-01.fw' is invalid
```

What cause my problem?