ljalves/linux_media #165: Problem building drivers on recent kernel

dbrb2, 2017-10-10:
Hi,
I have a machine that was running 4.4 kernel, and is now running 4.13. Both experienced the problem below: 

When trying to build the drievrs from source, all seems to be fine until:
 sudo modprobe saa716x_budget int_type=1 
modprobe: FATAL: Module saa716x_budget not found in directory /lib/modules/4.13.5-041305-generic

I have both a dual DVB-T tuner and a DVB-S tuner: 

02:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 02)
03:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)

Any suggestions? Would reverting to a particular kernel version help...?

THN1, 2017-11-13:
Hi

I am running a TechniSat SkyStar 2 eXpress HD card with .

It has compiled well with Luis' sources up to Kernel 4.10 including.

As of Kernel 4.11 and higher (eg Ubuntu 17.10) it seems you need to
- reference to sched.h should be replaced by sched/signal.h
- calls to function pci_enable_msix should be replaced by calls to pci_enable_msix_range

Hope it helps.

T.

Zuikkis, 2017-11-26:
The saa716x_budget driver is not build.

I can't figure out why. I see the source code there but no .o is generated. I see no errors, imho it doesn't even try to build it.

Zuikkis, 2017-11-27:
saa716x_budget is not enabled in config.

Go to v4l directory, type "make config" and enable it. Then make and it does build it.

I first had to modify the Kconfig file slightly, "make config" complains about the "---help---" lines.. When replacing all of those with "help" it worked nicely.

manio, 2018-03-01:
> calls to function pci_enable_msix should be replaced by calls to pci_enable_msix_range

No, it has to be replaced with **pci_enable_msix_exact()**
**pci_enable_msix_range()** is returning positive value when success, so it would not work in this case without source code adjustment.

btw: @crazycat69 has a proper patch for this: https://github.com/crazycat69/linux_media/commit/79e0a1c5121ad34186d3f21d151b128905721a89#diff-f5e77d09edeb6d8c2e8f266122302a2c

FWIW: I successfully compiled this ljalves drivers on kernel 4.16-rc2, but I stripped the modules only to support single card relevant for me (TBS 6982SE) and I've putted it in the kernel tree directly, not compiling out of tree. Works great! :)

THN1, 2018-03-05:
Hi Mariusz 
thanks a lot for the clarification and for the patch provided. Appreciated.
Best regards