ljalves/linux_media #166: TBS 6985 on Ubuntu Server 17.10 - Output from Kernel very strange an only HD will work. SD not.

svalcken, 2017-11-04:
Here are the Kernel Output. Any Warnings and Traces, i have been found:

[    7.945230] media: loading out-of-tree module taints kernel.
[    7.945251] media: module verification failed: signature and/or required key missing - tainting kernel
[    7.945467] media: Linux media interface: v0.10
[    8.015597] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	cf382935d479868da551bf1593b3ee24afd5d54d fixed the problem that  tbs6285 can not get TS stream
[    8.200860] SAA716x Budget 0000:06:00.0: enabling device (0100 -> 0102)
[    8.228093] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[    8.228271] i2c i2c-9: Added multiplexed i2c bus 10
[    8.228297] i2c i2c-9: Added multiplexed i2c bus 11
[    8.612664] i2c i2c-11: av201x: Airoha Technology AV201x successfully attached
[    8.612960] SAA716x Budget 0000:06:00.0: TurboSight TBS 6985  MAC=00:22:ab:90:45:6c
[    8.612962] SAA716x Budget 0000:06:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6985 DVB-S/S2)...
[    8.613041] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[    8.613196] sysfs: cannot create duplicate filename '/devices/pci0000:00/0000:00:1c.6/0000:06:00.0/i2c-9/channel-0'
[    8.613202] ------------[ cut here ]------------
[    8.613206] WARNING: CPU: 2 PID: 367 at /build/linux-XO_uEE/linux-4.13.0/fs/sysfs/dir.c:31 sysfs_warn_dup+0x56/0x70
[    8.613206] Modules linked in: av201x(OE) saa716x_tbs_dvb(OE+) tas2101(OE) i2c_mux cx24117(OE) saa716x_core(OE) dvb_core(OE) media(OE) sunrpc parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_generic usbhid hid i915 i2c_algo_bit drm_kms_helper ahci syscopyarea libahci sysfillrect sysimgblt fb_sys_fops r8169 drm mii wmi video
[    8.613218] CPU: 2 PID: 367 Comm: systemd-modules Tainted: G           OE   4.13.0-16-generic #19-Ubuntu
[    8.613219] Hardware name: System manufacturer System Product Name/P8Z77-V LX, BIOS 2501 07/21/2014
[    8.613220] task: ffff9dda0b39c5c0 task.stack: ffffbfe3011fc000
[    8.613221] RIP: 0010:sysfs_warn_dup+0x56/0x70
[    8.613221] RSP: 0018:ffffbfe3011ff918 EFLAGS: 00010286
[    8.613222] RAX: 0000000000000067 RBX: ffff9dda103ec000 RCX: 0000000000000006
[    8.613223] RDX: 0000000000000000 RSI: 0000000000000092 RDI: ffff9dda1fb0dc70
[    8.613223] RBP: ffffbfe3011ff930 R08: 0000000000000001 R09: 000000000000035e
[    8.613224] R10: 0000000000000001 R11: 0000000000000000 R12: ffffbfe3011ff99c
[    8.613224] R13: ffff9dda0ad74708 R14: 0000000000000001 R15: ffffffffffffffef
[    8.613225] FS:  00007f080d502940(0000) GS:ffff9dda1fb00000(0000) knlGS:0000000000000000
[    8.613225] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    8.613226] CR2: 00007ff7b5b81bff CR3: 000000021097a000 CR4: 00000000001406e0
[    8.613227] Call Trace:
[    8.613230]  sysfs_do_create_link_sd.isra.2+0x9f/0xb0
[    8.613232]  sysfs_create_link+0x25/0x40
[    8.613233]  i2c_mux_add_adapter+0x216/0x9f0 [i2c_mux]
[    8.613235]  tas2101_attach+0xa6/0x970 [tas2101]
[    8.613238]  saa716x_tbs6985_frontend_attach+0x7a/0x2b0 [saa716x_tbs_dvb]
[    8.613241]  saa716x_dvb_init+0x260/0x560 [saa716x_core]
[    8.613243]  saa716x_budget_pci_probe+0x1a5/0x250 [saa716x_tbs_dvb]
[    8.613245]  local_pci_probe+0x42/0xa0
[    8.613247]  pci_device_probe+0x18d/0x1a0
[    8.613248]  driver_probe_device+0x2ff/0x450
[    8.613249]  __driver_attach+0xa4/0xe0
[    8.613250]  ? driver_probe_device+0x450/0x450
[    8.613252]  bus_for_each_dev+0x6e/0xb0
[    8.613253]  driver_attach+0x1e/0x20
[    8.613254]  bus_add_driver+0x1c7/0x270
[    8.613255]  ? 0xffffffffc0514000
[    8.613256]  driver_register+0x60/0xe0
[    8.613256]  ? 0xffffffffc0514000
[    8.613257]  __pci_register_driver+0x4c/0x50
[    8.613259]  saa716x_budget_pci_driver_init+0x1e/0x1000 [saa716x_tbs_dvb]
[    8.613261]  do_one_initcall+0x50/0x1a0
[    8.613263]  ? __vunmap+0x81/0xb0
[    8.613265]  ? kfree+0x14a/0x160
[    8.613266]  ? kmem_cache_alloc_trace+0x150/0x1b0
[    8.613267]  ? do_init_module+0x27/0x1f9
[    8.613269]  do_init_module+0x5f/0x1f9
[    8.613270]  load_module+0x183b/0x1f20
[    8.613272]  ? ima_post_read_file+0x7e/0xa0
[    8.613273]  SYSC_finit_module+0xfc/0x120
[    8.613274]  ? SYSC_finit_module+0xfc/0x120
[    8.613276]  SyS_finit_module+0xe/0x10
[    8.613277]  entry_SYSCALL_64_fastpath+0x1e/0xa9
[    8.613278] RIP: 0033:0x7f080d0339f9
[... 495 lines cut ...]
[    8.613272]  ? ima_post_read_file+0x7e/0xa0
[    8.613273]  SYSC_finit_module+0xfc/0x120
[    8.613274]  ? SYSC_finit_module+0xfc/0x120
[    8.613276]  SyS_finit_module+0xe/0x10
[    8.613277]  entry_SYSCALL_64_fastpath+0x1e/0xa9
[    8.613278] RIP: 0033:0x7f080d0339f9
[    8.613278] RSP: 002b:00007fffab6f7fe8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    8.613279] RAX: ffffffffffffffda RBX: 00007fffab6f6fe0 RCX: 00007f080d0339f9
[    8.613280] RDX: 0000000000000000 RSI: 000055eaba696f60 RDI: 0000000000000005
[    8.613280] RBP: 0000000000000005 R08: 0000000000000000 R09: 000055eab8a99099
[    8.613281] R10: 0000000000000005 R11: 0000000000000246 R12: 000055eaba69d900
[    8.613281] R13: 0000000000000005 R14: 00007fffab6f6fc0 R15: 0000000000000005
[    8.613282] Code: 85 c0 48 89 c3 74 12 b9 00 10 00 00 48 89 c2 31 f6 4c 89 ef e8 0c c8 ff ff 4c 89 e2 48 89 de 48 c7 c7 d0 9f cc a1 e8 cb 78 e0 ff <0f> ff 48 89 df e8 60 bd f4 ff 5b 41 5c 41 5d 5d c3 66 0f 1f 84 
[    8.613298] ---[ end trace 902f2e77ca28e8f5 ]---
[    8.613298] can't create symlink for channel 0"

Thirth:
I am not sure, i this an failure but i think so:
"[   10.208732] ACPI Warning: SystemIO range 0x0000000000000428-0x000000000000042F conflicts with OpRegion 0x0000000000000400-0x000000000000047F (\PMIO) (20170531/utaddress-247)
[   10.208736] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver"

Fourth:
How can i fix the SD Channels?

Befor Ubuntu Server 17.10 i have used the Ubuntu Desktop 16.04 with the TBS-Drivers. This has been worked fine for me, but on Version 17.10 the drivers won't compile. Any errors and only 2 Errors i have found to fix. i won't reinstall with older Linux.

THN1, 2017-11-13:
I'm not sure you can compile Luis's sources for a Kernel higher that 4.10.
It seems you first need some patches, please check.
T.

Krobar, 2017-11-21:
Are the duplicate I2C errors part of the problem? Are the patches mentioned above available yet? 

I seem to get similar errors running on 4.13.12-1-ARCH:
[    3.107688] ixgbevf: Copyright (c) 2009 - 2015 Intel Corporation.
[    3.111760] WARNING: You are using an experimental version of the media stack.
[    3.158608] ixgbevf 0000:00:03.0: 00:1e:67:65:93:05
[    3.158654] ixgbevf 0000:00:03.0: MAC: 2
[    3.158658] ixgbevf 0000:00:03.0: Intel(R) X540 Virtual Function
[    3.158709] saa716x_pci_init (0): found a TurboSight TBS 6985  PCIe card
[    3.160895] ACPI: PCI Interrupt Link [GSIE] enabled at IRQ 20
[    3.169308]     SAA7160 Rev 2 [6985:0002],
[    3.169312] irq: 35,
[    3.169315]     SAA7160 64Bit, MSI Enabled, MSI-X=32 msgs

[    3.194508] saa716x_i2c_hwinit (0): Adapter (b000) SAA716x I2C Core 0 RESET
[    3.203205] saa716x_i2c_hwinit (0): Adapter (c000) SAA716x I2C Core 1 RESET
[    3.208161] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[    3.214789] i2c i2c-2: Added multiplexed i2c bus 3
[    3.214847] i2c i2c-2: Added multiplexed i2c bus 4
[    3.625633] input: PC Speaker as /devices/platform/pcspkr/input/input4
[    3.626942] iTCO_vendor_support: vendor-support=0
[    3.629195] iTCO_wdt: Intel TCO WatchDog Timer Driver v1.11
[    3.629474] iTCO_wdt: Found a ICH9 TCO device (Version=2, TCOBASE=0x0660)
[    3.630099] iTCO_wdt: initialized. heartbeat=30 sec (nowayout=0)
[    3.669570] i2c i2c-4: av201x: Airoha Technology AV201x successfully attached
[    3.669882] SAA716x Budget 0000:00:04.0: TurboSight TBS 6985  MAC=00:22:ab:90:7e:88
[    3.669893] SAA716x Budget 0000:00:04.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6985 DVB-S/S2)...
[    3.672458] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[    3.683968] ppdev: user-space parallel port driver
[    3.684113] sysfs: cannot create duplicate filename '/devices/pci0000:00/0000:00:04.0/i2c-2/channel-0'
[    3.684171] ------------[ cut here ]------------
[    3.684185] WARNING: CPU: 0 PID: 215 at fs/sysfs/dir.c:31 sysfs_warn_dup+0x56/0x70
[    3.684188] Modules linked in: evdev(+) ppdev av201x(O) mac_hid iTCO_wdt iTCO_vendor_support pcspkr saa716x_tbs_dvb(O+) cx24117(O) saa716x_core(O) tas2101(O) drm dvb_core(O) i2c_mux ixgbevf parport_pc intel_agp syscopyarea sysfillrect sysimgblt qemu_fw_cfg intel_gtt fb_sys_fops lpc_ich parport media(O) shpchp i2c_i801 i2c_algo_bit agpgart button sch_fq_codel ip_tables x_tables ext4 crc16 mbcache jbd2 fscrypto sr_mod cdrom sd_mod virtio_balloon serio_raw atkbd libps2 ahci libahci libata scsi_mod xhci_pci xhci_hcd virtio_pci virtio_ring virtio usbcore usb_common i8042 serio
[    3.684551] CPU: 0 PID: 215 Comm: systemd-udevd Tainted: G           O    4.13.12-1-ARCH #1
[    3.684555] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 1.11.0-20171110_095948-anatol 04/01/2014
[    3.684559] task: ffff93587b558dc0 task.stack: ffffa507004e8000
[    3.684567] RIP: 0010:sysfs_warn_dup+0x56/0x70
[    3.684571] RSP: 0018:ffffa507004eb8e8 EFLAGS: 00010282
[    3.684577] RAX: 000000000000005a RBX: ffff93587cdcc000 RCX: 0000000000000000
[    3.684581] RDX: 0000000000000000 RSI: ffff93587f00dc78 RDI: ffff93587f00dc78
[    3.684584] RBP: ffffa507004eb900 R08: 0000000000000208 R09: 0000000000000004
[    3.684587] R10: 0000000000000001 R11: 0000000000000001 R12: ffffa507004eb96c
[    3.684591] R13: ffff93587cee9708 R14: ffff93587cee9708 R15: ffffffffffffffef
[    3.684604] FS:  00007fab2f85c0c0(0000) GS:ffff93587f000000(0000) knlGS:0000000000000000
[    3.684608] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    3.684612] CR2: 00007fab2aecb6f8 CR3: 000000003b562000 CR4: 00000000000006f0
[    3.684624] Call Trace:
[    3.684639]  sysfs_do_create_link_sd.isra.2+0xbc/0xd0
[    3.684649]  sysfs_create_link+0x25/0x40
[    3.684660]  i2c_mux_add_adapter+0x216/0x9f0 [i2c_mux]
[    3.684674]  tas2101_attach+0xa6/0x970 [tas2101]
[    3.684694]  saa716x_tbs6985_frontend_attach+0x7a/0x2b0 [saa716x_tbs_dvb]
[    3.684707]  saa716x_dvb_init+0x260/0x560 [saa716x_core]
[    3.684725]  saa716x_budget_pci_probe+0x1a5/0x250 [saa716x_tbs_dvb]
[    3.684735]  local_pci_probe+0x42/0xa0
[    3.684741]  ? pci_match_device+0xe0/0x110
[    3.684749]  pci_device_probe+0x18d/0x1a0
[    3.684760]  driver_probe_device+0x2ff/0x450
[    3.684769]  __driver_attach+0xa4/0xe0
[    3.684776]  ? driver_probe_device+0x450/0x450
[    3.684784]  bus_for_each_dev+0x6e/0xb0
[    3.684793]  driver_attach+0x1e/0x20
[    3.684799]  bus_add_driver+0x1c7/0x270
[    3.684805]  ? 0xffffffffc055d000
[    3.684812]  driver_register+0x60/0xe0
[    3.684818]  ? 0xffffffffc055d000
[    3.684824]  __pci_register_driver+0x4c/0x50
[    3.684838]  saa716x_budget_pci_driver_init+0x1e/0x1000 [saa716x_tbs_dvb]
[    3.684846]  do_one_initcall+0x50/0x190
[    3.684854]  ? kfree+0x175/0x190
[... 54 lines cut ...]
[    3.685570]  saa716x_budget_pci_driver_init+0x1e/0x1000 [saa716x_tbs_dvb]
[    3.685577]  do_one_initcall+0x50/0x190
[    3.685584]  ? kfree+0x175/0x190
[    3.685588]  ? kfree+0x175/0x190
[    3.685597]  ? do_init_module+0x27/0x1f2
[    3.685607]  do_init_module+0x5f/0x1f2
[    3.685617]  load_module+0x23f3/0x2be0
[    3.685631]  ? vfs_read+0x115/0x130
[    3.685648]  SYSC_finit_module+0xfc/0x120
[    3.685655]  ? SYSC_finit_module+0xfc/0x120
[    3.685672]  SyS_finit_module+0xe/0x10
[    3.685678]  do_syscall_64+0x54/0xc0
[    3.685686]  entry_SYSCALL64_slow_path+0x25/0x25
[    3.685691] RIP: 0033:0x7fab2f167e79
[... cut ...]
[    3.685715] R13: 000055ef6b30c7b0 R14: 0000000000020000 R15: 0000000000000000
[    3.685727] Code: cd 44 03 fa 8b 45 b0 eb 9c 48 c7 c7 b8 61 45 c0 e8 5d 9f c8 f9 0f ff e9 6f fe ff ff 44 89 ee 48 c7 c7 e0 61 45 c0 e8 47 9f c8 f9 <0f> ff e9 8d fe ff ff 48 8b 7f 08 48 c7 c6 e8 60 45 c0 e8 8f 44
[    3.685846] ---[ end trace cd1ea7d1dccf46e6 ]---
[    3.685854] i2c i2c-2: Added multiplexed i2c bus 5
[    3.685979] sysfs: cannot create duplicate filename '/devices/pci0000:00/0000:00:04.0/i2c-2/channel-1'
[    3.686013] ------------[ cut here ]------------

closed 2024-03-09 by svalcken