ljalves/linux_media #171: make is failing after upgrade to Ubuntu 18.04 - Please don't include <linux/compiler-gcc.h> directly,

waxhead, 2018-08-25:
Hi,

After upgrading my myth back end to Ubuntu 18.04 I tried to compile the drivers for the TBS tuner card and get this error:

```
Syncing with dir ../../media
make[2]: Leaving directory '/home/pete/os-tbs/media_build/linux'
make -C /lib/modules/4.15.0-33-generic/build SUBDIRS=/home/pete/os-tbs/media_build/v4l  modules
make[2]: Entering directory '/usr/src/linux-headers-4.15.0-33-generic'
  CC [M]  /home/pete/os-tbs/media_build/v4l/cec-edid.o
In file included from ./include/linux/compiler_types.h:58:0,
                 from ././include/linux/kconfig.h:74,
                 from <command-line>:0:
/home/pete/os-tbs/media_build/v4l/../linux/include/linux/compiler-gcc.h:2:2: error: #error "Please don't include <linux/compiler-gcc.h> directly, include <linux/compiler.h> instead."
 #error "Please don't include <linux/compiler-gcc.h> directly, include <linux/compiler.h> instead."
  ^~~~~
scripts/Makefile.build:339: recipe for target '/home/pete/os-tbs/media_build/v4l/cec-edid.o' failed
make[3]: *** [/home/pete/os-tbs/media_build/v4l/cec-edid.o] Error 1
Makefile:1552: recipe for target '_module_/home/pete/os-tbs/media_build/v4l' failed
make[2]: *** [_module_/home/pete/os-tbs/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.15.0-33-generic'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/home/pete/os-tbs/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2

```

Googling shows this is problem for other applicaitons, but I'm not sure how to address this.

Thanks.

crazycat69, 2018-08-25:
https://github.com/tbsdtv/media_build

waxhead, 2018-08-25:
Thanks, that worked perfectly!!! 

:+1:

closed 2018-08-25 by waxhead