ljalves/linux_media #175: TBS6205 (quad T/T2) and TBS6902 (dual S/S2) persistent config

grandmastermarclar, 2020-01-03:
I've got both cards above. The 6902 replaces a 6982SE which appears to be dead. Because they used different drivers, I could use the adapter_nr trick to assign the old S/S2 card to be adapter4 and adapter5.

The 6205 and 6902 both use tbsecp3, so I guess I can't do that. How can I, under Ubuntu 18.04, keep the 6205 as adapter0-3, and the 6902 as adapter4-5.. or somehow set up aliases to do this?

TIA.

crazycat69, 2020-01-03:
Both cards handled same driver, so adapter order always same (depend from num on pci bus). No need adapter_nr  option, maybe need change cards btw pci-e slots.

grandmastermarclar, 2020-01-04:
Swapped the cards over; and I get the 6205 as adapter0-3, and the 6902 as adapter4-5. Exactly as I had wanted, persistent too. However, we should be long past the days when we have to hope for good fortune like this. There really should (i.e. must) be a way where tuners can be assigned as required by the user, not by the luck of how one card is initialised before another which could easily change and therefore put me back to my original question.

Consider this closed until some other kernel jiggery-pokery changes this fortunately default behaviour.

closed 2020-01-04 by grandmastermarclar