ljalves/linux_media #180: TBS6991 support

woooow1, 2024-05-16:
Hello Luis, 
i need some support or advice for this card! I am trying to run it in opensource driver, but with no luck so far!
In code I see that : 

tbs6991 seems to have different settings depending on a value that the closed
source driver reads from the eeprom. This is probabbly related to card HW revisons.

When I change default TBS6991_TSMODE0 to 1, card was able to tune in tvheadend, but with lot of errors. When I change to 0- unable to tune! Transport error indicator (total 1)
I also try do decompile closed source driver to find any useful info.This is related to tsmode: 

` tbs6991fe_wr(state_00,0xd,uVar2 | 0xc0);
    if (state_00->demod == 0) {
    tbs6991fe_wr(state_00,0xe6,0x10);
    tbs6991fe_wr(state_00,0xe7,0x32);
    tbs6991fe_wr(state_00,0xe8,0x54);
    tbs6991fe_wr(state_00,0xe9,0x76);
    tbs6991fe_wr(state_00,0xea,0xa8);
    tbs6991fe_wr(state_00,0xeb,0x9b);
    iVar1 = state_00->tsmode;
  }
  else {
    tbs6991fe_wr(state_00,0xe6,0x30);
    tbs6991fe_wr(state_00,0xe7,0x21);
    tbs6991fe_wr(state_00,0xe8,0x54);
    tbs6991fe_wr(state_00,0xe9,0x76);
    tbs6991fe_wr(state_00,0xea,0xb8);
    tbs6991fe_wr(state_00,0xeb,0x9a);
    iVar1 = state_00->tsmode;
  }
  if (iVar1 == 1) {
    tbs6991fe_wr(state_00,0xe0,0x31);
  }
  else {
    tbs6991fe_wr(state_00,0xe0,0x33);
  }`
Any help is welcome!

crazycat69, 2024-05-17:
This demod TS bus config (TAS2101 pins for TS bus).
Need configure CI support chip for different TBS cards - use TBS fork which contain CI support for SAA716x based cards
https://github.com/tbsdtv/linux_media
https://github.com/tbsdtv/media_build

woooow1, 2024-05-18:
@crazycat69 , thanks for reply! I already use tbs fork, but card not work! Tested also on fta channels. Card is v13!

crazycat69, 2024-05-18:
TBS6991_TSMODE0 (0x33) is parallel TS
TBS6991_TSMODE1 (0x31) is serial TS

crazycat69, 2024-05-18:
you have **"tbsci: Initializing TBS 6991 v13 CI"** in kernel log ?

crazycat69, 2024-05-18:
https://github.com/tbsdtv/linux_media/blob/c561bd051b09cc16622d021713b75f1b814d9e19/drivers/media/pci/saa716x/saa716x_budget.c#L2487
change to 
       ret = tbsci_i2c_probe(adapter,count?3:4);

woooow1, 2024-05-19:
> you have **"tbsci: Initializing TBS 6991 v13 CI"** in kernel log ?

 Yes, tbsci: Initializing TBS 6991 v13 CI 1 slot
I already try you suggestion, with  ret = tbsci_i2c_probe(adapter,count?3:4); no ci slot is initialized!

woooow1, 2025-03-28:
https://github.com/tbsdtv/linux_media/commit/b68595ce48c1ebfc983e1117527b4a96125b5d25

closed 2025-03-28 by woooow1